tomas789/kitti2bag #7: Camera info height and width swapped

Humhu, 2017-06-12:
I have observed that the output sensor_msgs/CameraInfo messages have their height and width fields flipped, while the sensor_msgs/Image messages do not. 

I am running OpenCV 3 and with the latest version of kitti2bag, and have not tested if this issue occurs with OpenCV 2.x.

tomas789, 2017-06-12:
Thank you for bug report. I'll fix it soon.

closed 2017-07-12 by tomas789

quhezheng, 2017-11-23:
@tomas789 Please respect your user, we use this tool because it is recommended on KITTI.
But the issue is not fixed at all. This is github, if don't want to open source your tool. Please close this repository. Because we can't fix the issue locally.

BmlmnnsDev, 2019-02-01:
I am running latest version of kitti2bag. Problem seems to be still existent.

lixz123007, 2019-07-22:
it is still existent

sreeharshaparuchur1, 2021-02-06:
Hey, so you can solve this issue by cloning kitti2bag in your catkin/src workspace. Build the ws (preferably using `catkin build`) and source the setup file. Now, navigate to `./catkin_ws/src/kitti2bag/kitti2bag`, you'll find a `__main__.py` script. In that, change the first line from `from  .kitti2bag import run_kitti2bag` to `from kitti2bag import run_kitti2bag`. Change the line in the kitti2bag.py file (swapping height and width, [line number 117](https://github.com/tomas789/kitti2bag/blob/master/kitti2bag/kitti2bag.py#L117)). Now, assuming that you're using the same drive as provided in the example (the raw data -take care that it's not the rectified images as then you'll end up rectifying a rectified image) , have all the required dependencies installed (opencv, scikit-build) and the data (calib and sync) in the correct ordering, run the command: `python2 __main__.py raw_synced -t  2011_09_26 -r 0002`. Ensure that you use python2 and not python3 as ROS runs on python2. Using Python3 may result in errors like `ImportError: dynamic module does not define module export function (PyInit__tf2)`.

Hope this helps