tomas789/kitti2bag #9: Segmentation fault (core dumped)

Mikhail2017, 2017-06-29:
Hi.
I'm trying to use your converter.
Ubuntu 16.04. 
Python 2.7.12 & 3.5.2
ROS installed

kitti2bag was installed using pip

Executing next string
kitti2bag -t 2011_09_26 -r 0001 raw_synced .

always leads to

Segmentation fault (core dumped)

tomas789, 2017-07-12:
Hello, it looks like you have a problem with your ROS/python. I'll try to reproduce it but I've seen this before and I fixed it by reinstalling pykitti and its dependencies. Also make sure that you have ROS installed correctly.

fencingworld, 2017-07-21:
I have met the same problem "Segmentation fault (core dumped)",which is caused by python interpreter in my computer.
There are two python interpreters in my computer,one is ubuntu's and the other is anaconda. 
The rospy used by kitti2bag is only supported by the  one of ubuntu.
Finally I manage it with the ubuntu python interpreter.

tomas789, 2017-10-21:
I will close this issue since it likely was conflict of different versions. On [CI](https://travis-ci.org/tomas789/kitti2bag), everything works as expected. Feel free to reopen this issue when you'll experience this issue again.

closed 2017-10-21 by tomas789