tomas789/kitti2bag #10: Unable to generate rosbag from the KITTI odometry Velodyne dataset

vkee, 2017-07-12:
Hi,

I'm unable to generate a rosbag from the KITTI odometry Velodyne dataset (http://www.cvlibs.net/download.php?file=data_odometry_velodyne.zip).

I get a AttributeError: odometry instance has no attribute 'load_calib' when I try to run kitti2bag odom_color -s 01.

Has anyone been able to do this successfully?

Thanks!

tomas789, 2017-07-12:
I'm currently downloading the dataset to try it but it will take a while since it is very large. This is probably caused by `pykitti` since they changed API. It should be already fixed. Can you try it with current version please? Try the version from Github although newest version from pip should work too.

vkee, 2017-07-12:
I installed everything today and had this issue. Please let me know if you are able to get it to work.

Thanks!

vkee, 2017-07-24:
Any updates on this? I tried both the latest version from Github along with the latest from pip.

Thanks!

vkee, 2017-07-24:
I was able to generate the bag with the raw dataset but I noticed that there are some issues with the tf generated by this tool. It is not consistent with what is in the figure from the paper
[image]

As you can see in a screenshot I made from playing the bag generated of 2011_09_30_drive_0020, the cameras are not in the correct poses. Additionally, it appears that the left and right cameras are flipped, as well as their relative positions (order from left to right is 1, 3, 0, 2 instead of 2, 0, 3, 1). The displacement between the GPS/IMU and the velodyne along the y axis should be 0.32 but it is over a centimeter off while the calibration file was correct).
[image]

bartville, 2017-09-06:
Hi guys, I've got the same exactly issue while converting a sequence of kitti velodyne datasets

> AttributeError: odometry instance has no attribute 'load_calib'

Any updates?

CtfChan, 2017-09-07:
Hey bartville I am having the exact same issue as you with the stereo dataset. kitti2bag calls on nonexistent functions from pykitti and the script just crashes.

AbnerCSZ, 2017-12-20:
I have encountered the same problem as yours.#16  And I solved it. The odometry datasets are just subsets of the raw data. We can't  use kitti2bag to directly convert odometry data to rosbag. But we can find the mapping from odometry sequences to raw sequences in the readme.txt file included with the odometry development kit, and then use kitti2bag.

> In order to allow the usage of the laser point clouds, gps data, the right
> camera image and the grayscale images for the TRAINING data as well, we
> provide the mapping of the training set to the raw data of the KITTI dataset.
> The following table lists the name, start and end frame of each sequence that
> has been used to extract the visual odometry / SLAM training set:
> 
> Nr.     Sequence name     Start   End
> ---------------------------------------
> 00: 2011_10_03_drive_0027 000000 004540
> 01: 2011_10_03_drive_0042 000000 001100
> 02: 2011_10_03_drive_0034 000000 004660
> 03: 2011_09_26_drive_0067 000000 000800
> 04: 2011_09_30_drive_0016 000000 000270
> 05: 2011_09_30_drive_0018 000000 002760
> 06: 2011_09_30_drive_0020 000000 001100
> 07: 2011_09_30_drive_0027 000000 001100
> 08: 2011_09_30_drive_0028 001100 005170
> 09: 2011_09_30_drive_0033 000000 001590
> 10: 2011_09_30_drive_0034 000000 001200

wutongyu98, 2018-10-31:
modified version of *https://github.com/tomas789/kitti2bag*
1. Install pykitti pip install pykitti

2. Modify pykitti/odometry.py file located in /usr/local/lib/python2.7/dist-packages/pykitti. At the end of the odometry.py, add:

   def load_poses(self):
        """Load ground truth poses from file."""
        print('Loading poses for sequence ' + self.sequence + '...')
 
        pose_file = os.path.join(self.pose_path, self.sequence + '.txt')
 
        # Read and parse the poses
        self.T_w_cam0 = []
        with open(pose_file, 'r') as f:
             for line in f.readlines():
                 T = np.fromstring(line, dtype=float, sep=' ')
                 T = T.reshape(3, 4)
                 T = np.vstack((T, [0, 0, 0, 1]))
                 self.T_w_cam0.append(T)
        print('done.')
3. Install kitti2bag: pip install kitti2bag

4. Modify kitti2bag located in /usr/local/bin/. Change from:

	kitti.load_calib()
        kitti.load_timestamps()
To:

        kitti._load_calib()
        kitti._load_timestamps()
5. Enter the dataset folder with image sequence and pose gt. e.g.: enter dataset/, create poses (00.txt, 01.txt, ...), and /sequences (00/image_0, 00/image_1, 00/calib.txt, 00/times.txt).

6. Enter kitti2bag -s 00 odom_gray . to convert 00 bag to rosbag

getupgetup, 2018-11-16:
> I have encountered the same problem as yours.#16 And I solved it. The odometry datasets are just subsets of the raw data. We can't use kitti2bag to directly convert odometry data to rosbag. But we [...]

 But this means I need to download raw data again which contains large amount of images that I don't need.Is there anyway to get rosbag from the pure velodyne data sequence?

HongweiSunny, 2018-12-21:
> But this means I need to download raw data again which contains large amount of images that I don't need.Is there anyway to get rosbag from the pure velodyne data sequence?

@getupgetup  Hi, is this probelm solved?

getupgetup, 2018-12-21:
> @getupgetup Hi, is this probelm solved?

没有,不过一个可行的方案:参考楼上的方法修改kitti2bag代码,然后把raw data下载后转换为rosbag,如果不需要图像等其他数据,就在kitti2bag源码中注释掉不需要的部分.

AbnerCSZ, 2018-12-21:
@getupgetup @HongweiSunny 
If you only want to get rosbag from velodyne, you can just download the velodyne .bin data and run this code:
```
 ros::init(argc, argv, "lidar2rosbag");
  ros::NodeHandle nh;

  if(argc<3)
  {
      printf("ERROR: Please follow the example: rosrun pkg node input num_output:\n  rosrun loam_velodyne lidar2rosbag /data/KITTI/dataset/sequences/04/ 04 \n");
      return -2;

  }
     

  pubLaserCloud = nh.advertise<sensor_msgs::PointCloud2>("/velodyne_points_pub", 2);

 
    std::string input_dir = argv[1];
    std::string output_dir = argv[2];
    std::string bin_path = input_dir + "velodyne/" ;//"/data/KITTI/dataset/sequences/04/velodyne/";
    std::string times_path = input_dir + "times.txt";

    //load times
    times_lists.clear();
    ifstream timeFile(times_path, std::ios::in);
  
    if(timeFile.is_open())
    {
        double time = 0.0f;
        while(!timeFile.eof() )
        {
            
            timeFile >>setprecision(12)>> time;
            times_lists.push_back(time);
        }
    }

    read_filelists( bin_path, file_lists, "bin" );
    sort_filelists( file_lists, "bin" );
    
    for(int i =0;i<file_lists.size();i++)
    {
        std::cout << file_lists[i]<<std::endl;
    }

    for(int i =0;i<times_lists.size();i++)
    {
        std::cout << times_lists[i]<<std::endl;
    }

    std::cout <<"size:  "<< file_lists.size()<<"    "<<times_lists.size()<<endl;
    
    rosbag::Bag bag;
    bag.open(output_dir + ".bag", rosbag::bagmode::Write);

    //load point cloud
    for(int iter=0;iter<file_lists.size();iter++)
    {
        std::string infile = bin_path + file_lists[iter];
        ifstream input(infile.c_str(), ios::in | ios::binary);
        if(!input.is_open() ){
            cerr << "Could not read file: " << infile << endl;
            return -1;
        }
        // pcl::PointCloud<PointXYZI>::Ptr points (new pcl::PointCloud<PointXYZI>);
        pcl::PointCloud<pcl::PointXYZI> points;
        const size_t kMaxNumberOfPoints = 1e6;  // From Readme for raw files.
        points.clear();
        points.reserve(kMaxNumberOfPoints);

        int i;
        for (i=0; input.is_open() && !input.eof(); i++) {
            PointXYZI point;
            
            input.read((char *) &point.x, 3*sizeof(float));
            input.read((char *) &point.intensity, sizeof(float));
            points.push_back(point);
        }
        input.close();

        ros::Time timestamp_ros(times_lists[iter]==0?  ros::TIME_MIN.toSec()  :times_lists[iter]);
       // timestampToRos(times_lists[iter], &timestamp_ros);

        points.header.stamp = times_lists[iter] ;
        points.header.frame_id = "velodyne";

        sensor_msgs::PointCloud2 output;
        pcl::toROSMsg(points, output);
        output.header.stamp = timestamp_ros  ;
        output.header.frame_id = "velodyne";
        
        pubLaserCloud.publish(output);
        std::cout<<"ros time : "<< output.header.stamp.toSec() <<"  with  "<<timestamp_ros.toSec()<<endl;
        bag.write("velodyne_points", timestamp_ros, output);

    }

    printf("lidar 2 kitti rosbag done\n");
```

getupgetup, 2018-12-24:
> @getupgetup @HongweiSunny
> If you only want to get rosbag from velodyne, you can just download the velodyne .bin data and use my tool: [lidar2rosbag_KITTI](https://github.com/AbnerCSZ/lidar2rosbag_KITTI) .

还没细看,谢谢~

HongweiSunny, 2018-12-24:
> @getupgetup @HongweiSunny
> If you only want to get rosbag from velodyne, you can just download the velodyne .bin data and use my tool: [lidar2rosbag_KITTI](https://github.com/AbnerCSZ/lidar2rosbag_KITTI) .

Thanks for your replying!
@getupgetup  可否加个微信之类的平时可以交流一下  wechatID：sunhongwei0930

pangchenglin, 2019-01-28:
> > @getupgetup @HongweiSunny
> > If you only want to get rosbag from velodyne, you can just download the velodyne .bin data and use my tool: [lidar2rosbag_KITTI](https://github.com/AbnerCSZ/lidar2rosbag_KITTI) .
> 
> Thanks for your replying!
> @getupgetup 可否加个微信之类的平时可以交流一下 wechatID：sunhongwei0930

Are you fixed this problem? if you can fix it ,could you please tell me how to fix it???

nobinov, 2019-03-27:
> modified version of _https://github.com/tomas789/kitti2bag_
>
>     1. Install pykitti pip install pykitti
>
>     2. Modify pykitti/odometry.py file located in [...]

I've tried this, and this error occured : 

`AttributeError: odometry instance has no attribute 'load_poses'`

So, I changed `kitti.load_poses()` to `kitti._load_poses()` . After that, it give output like this : 
`AttributeError: odometry instance has no attribute 'T_w_cam0'`

Anyone knows what `'T_w_cam0'` is?

getupgetup, 2019-04-03:
> > modified version of _https://github.com/tomas789/kitti2bag_
> > ```
> > 1. Install pykitti pip install pykitti
> > 
> > 2. Modify pykitti/odometry.py file located in /usr/local/lib/python2.7/dist-packages/pykitti. At the end of the odometry.py, add:
> >    **def load_poses(self):
> >    "Load ground truth poses from file."
> >    print('Loading poses for sequence ' + self.sequence + '...')
> >    pose_file = os.path.join(self.pose_path, self.sequence + '.txt')
> >    # Read and parse the poses
> >    self.T_w_cam0 = []
> >    with open(pose_file, 'r') as f:
> >    for line in f.readlines():
> >    T = np.fromstring(line, dtype=float, sep=' ')
> >    T = T.reshape(3, 4)
> >    T = np.vstack((T, [0, 0, 0, 1]))
> >    self.T_w_cam0.append(T)
> >    print('done.')**
> > 
> > 3. Install kitti2bag: pip install kitti2bag
> > 
> > 4. Modify kitti2bag located in /usr/local/bin/. Change from:
> >    kitti.load_calib()
> >    kitti.load_timestamps()
> >    To:
> >    ```
> >     kitti._load_calib()
> >     kitti._load_timestamps()
> >    ```
> > 
> > 5. Enter the dataset folder with image sequence and pose gt. e.g.: enter dataset/, create poses (00.txt, 01.txt, ...), and /sequences (00/image_0, 00/image_1, 00/calib.txt, 00/times.txt).
> > 
> > 6. Enter kitti2bag -s 00 odom_gray . to convert 00 bag to rosbag
> > ```
> 
> I've tried this, and this error occured :
> 
> `AttributeError: odometry instance has no attribute 'load_poses'`
> 
> So, I changed `kitti.load_poses()` to `kitti._load_poses()` . After that, it give output like this :
> `AttributeError: odometry instance has no attribute 'T_w_cam0'`
> 
> Anyone knows what `'T_w_cam0'` is?
> 
> **UPDATE:**
> From example of pykitti for odometry data in [https://github.com/utiasSTARS/pykitti/blob/master/demos/demo_odometry.py](url) , `T_w_cam0`is a list of ground truth poses. so `kitti.T_w_cam0` can be replaced by `kitti.poses` . And now it's working.
> 
> n.b. : Apparently odom_color and odom_gray only didn't include velodyne data to the rosbag. Anyone knows how to include velodyne data to the bag?

I'm sorry that I've forgot how I solved the problem. I think the velodyne data with image/pose/etc data is included in the whole text file.Then different types of data are write to rosbag simultaneously by pykitti.
PS: Version incompatibility is really annoying.  I think you can learn some basic python which is not hard. Then you can modify code in pykitti according to error messages. (That is the way I have done to solve it)

chubukeji, 2022-01-20:
hello，the 03: 2011_09_26_drive_0067 can't be found in official website. Do you have this dataset? Or，do you know why it can't be found? my Email 2661076951@qq.com

chubukeji, 2022-01-20:
@AbnerCSZ  hello，the 03: 2011_09_26_drive_0067 can't be found in official website. Do you have this dataset? Or，do you know why it can't be found? my Email 2661076951@qq.com

Petros626, 2024-09-27:
@chubukeji I can't find it, so I assume it wasn't published.