tomas789/kitti2bag #11: tf data doesn't seem to exactly match up with gnd truth

vkee, 2017-08-01:
Hi,

I generated a bag file of sequence 6 of the KITTI odometry dataset from the raw data recordings using this tool. I then wrote a simple node that uses the rosbag tf to lookup the left camera coordinate pose at each frame and then stores this trajectory. I used the KITTI evaluation tool to plot and evaluate the trajectory.

[image]

As you can see, it is pretty close to the ground truth which is expected. However, I would expect it to be exactly the same or very near to it as the paper ([Are we ready for autonomous driving? the KITTI vision benchmark suite](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjj1IjGhbbVAhVNbVAKHRNhBPMQFggrMAA&url=http%3A%2F%2Fwww.cvlibs.net%2Fpublications%2FGeiger2012CVPR.pdf&usg=AFQjCNFnuL2zuUTg8iYw6kbW4wPDu-AvxQ)) says: "The ground truth for visual odometry/SLAM is directly given by the output of the GPS/IMU localization unit projected into the coordinate system of the left camera after rectification."

I did a pose by pose comparison and plotted the translation RMSE (distance between the poses) and for the most part it is close, but there is some significant error at some frames. I haven't been able to determine why there is that error as the ground truth is from the GPS/INS data directly and the tfs are based off of that.

[image]

Do you have any idea what may be the cause? 

Thanks,
Vincent

kerimyener, 2018-02-22:
@vkee hello would you mind to tell me how to convert sequence 6 of kitti odometry data to rosbag file i am a bit confused. Your help will be appreciated. Thank you

vkee, 2018-03-11:
@kerimyener I used raw datasets.

BrunoEduardoCSantos, 2019-02-14:
@vkee did you manage to solve the tf issue?

wxt-github, 2019-04-13:
Can you tell me where the sixth sequence is in the raw data, too much raw data, and no idea where it is. Also, I would like to ask if the sixth sequence is a complete closed loop? @vkee

valgur, 2019-05-14:
Related: https://github.com/ethz-asl/kitti_to_rosbag/issues/7

valgur, 2019-05-21:
@wxt-github The mapping from odometry to raw datasets is provided for the first 10 sequences in the readme of the [odometry devkit](https://s3.eu-central-1.amazonaws.com/avg-kitti/devkit_odometry.zip).

| Nr. | Sequence name         | Start  | End    |
|----:|-----------------------|--------|--------|
| 00  | 2011_10_03_drive_0027 | 000000 | 004540 |
| 01  | 2011_10_03_drive_0042 | 000000 | 001100 |
| 02  | 2011_10_03_drive_0034 | 000000 | 004660 |
| 03  | 2011_09_26_drive_0067 | 000000 | 000800 |
| 04  | 2011_09_30_drive_0016 | 000000 | 000270 |
| 05  | 2011_09_30_drive_0018 | 000000 | 002760 |
| 06  | 2011_09_30_drive_0020 | 000000 | 001100 |
| 07  | 2011_09_30_drive_0027 | 000000 | 001100 |
| 08  | 2011_09_30_drive_0028 | 001100 | 005170 |
| 09  | 2011_09_30_drive_0033 | 000000 | 001590 |
| 10  | 2011_09_30_drive_0034 | 000000 | 001200 |

hemangchawla, 2019-11-08:
Is this related to the camera rectification after the oxts to xyz?

J-Rojas, 2020-01-18:
I noticed some errors with the TF when I mapped the lidar data against the vehicle's motion over time. I believe the TFs are not calculated correctly because they are missing the transformation from IMU/GPS to the velodyne sensor. The calib_imu_to_velo.txt is not used in this code and that would most likely throw off any sensor that is correlating lidar/camera data to GPS coordinates.

chubukeji, 2022-01-20:
@valgur hello，the 03: 2011_09_26_drive_0067 can't be found in official website. Do you have this dataset?  Or，do you know why it can't be found? my Email 2661076951@qq.com

christian-hiebl, 2023-03-17:
@chubukeji This discussion suggests it was removed because parts overlapped with the public benchmark test.
https://github.com/alexkreimer/odometry/issues/1#issuecomment-534451019