tomas789/kitti2bag #12: What does /tf contain?

zanazakaryaie, 2017-11-21:
Hello

I'm not sure what does /tf message contains. Does it containt the pose of the left camera obtained with GPS/IMU fusion?

Can I use it as ground truth?

tomas789, 2017-12-10:
Hi! `/tf` contains relative transformations to all sensors you can find in dataset (including left camera). Besides that, it also contains position of car in the world. Thus you can use it as ground truth.

closed 2017-12-10 by tomas789

zanazakaryaie, 2017-12-10:
I have compared the tf with ground truth provided in odometry section of KITTI. They are different!

vkee, 2017-12-10:
Yes I have found that too, which is strange as it is based from the GPS/INS data (the source for ground truth). I tried contacting the authors but did not hear back.