tomas789/kitti2bag #15: No module named '_tf2'

richardzyx, 2017-12-15:
I'm running this package on Ubuntu 16.04 with `python` aliased to python3, I ran into this exception message:
```
Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 12, in <module>
    import tf
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 37, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 37, in <module>
    from _tf2 import *
ImportError: No module named '_tf2'

```
Is there some dependencies I'm missing? It seems to be related to this issue: https://github.com/ros/geometry2/issues/259 but I'm not sure. I've followed the tutorial on tf to install the packages.

Thanks!

breyner-posso, 2018-01-15:
Hi, I get the same error:

Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 12, in <module>
    import tf
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)

Did you solve this problem?

Tks a lot!

tomas789, 2018-01-15:
Hi @breyner-posso,

Can you check your environment? This error is almost certainly connected to your Python. Can you try running following command and sharing outputs? 

`python -c "import tf"`

`python2.7 -c "import tf"`

It will show if you can correctly import `tf` library. Similar error can also happen if you run it in virtualenv.

theplaystar, 2018-01-19:
I have the same problem
with

python2.7 -c "import tf"
Cannot load geometry_msgs moduleTraceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/__init__.py", line 47, in <module>
    from std_msgs.msg import Header
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/std_msgs/msg/__init__.py", line 1, in <module>
    from ._Bool import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/std_msgs/msg/_Bool.py", line 5, in <module>
    import genpy
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 40, in <module>
    import math
ImportError: No module named math
----
python -c "import tf"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)

tomas789, 2018-01-19:
@theplaystar According to your output, there is clearly problem with how your ROS setup, not kitti2bag. Problem is that your python does not see some parts of ROS. Can you double check that you correctly run `source /opt/ros/...` as recommended by ROS install guide? 

Since this issue is not caused by kitti2bag, I would recommend consulting [ROS Answers](https://answers.ros.org) or similar place.

GaoJunqiang, 2018-04-27:
sudo pip2 install kitti2bag
python2 kitti2bag -t 2011_09_29 -r 0071 raw_synced .
and bingo!

kplmcelligott, 2018-07-27:
I got it working using the following comands in ros-kinetic:

sudo python -m pip install --upgrade pip
sudo pip install kitti2bag
sudo apt-get update
sudo apt-get upgrade

closed 2018-07-31 by richardzyx