tomas789/kitti2bag #16: How can I only convert velodyne data to rosbag?

AbnerCSZ, 2017-12-19:
How can I only convert velodyne data (*.bin) to rosbag? Only need to convert velodyne data, without image data.
So how to use folder "velodyne_points" without calibration file or image folder?
Looking forward to any reply

AbnerCSZ, 2017-12-20:
My problem is solved.

> we provide the mapping of the training set to the raw data of the KITTI dataset.
> The following table lists the name, start and end frame of each sequence that
> has been used to extract the visual odometry / SLAM training set:
Nr.     Sequence name     Start   End
---------------------------------------
00: 2011_10_03_drive_0027 000000 004540
01: 2011_10_03_drive_0042 000000 001100
02: 2011_10_03_drive_0034 000000 004660
03: 2011_09_26_drive_0067 000000 000800
04: 2011_09_30_drive_0016 000000 000270
05: 2011_09_30_drive_0018 000000 002760
06: 2011_09_30_drive_0020 000000 001100
07: 2011_09_30_drive_0027 000000 001100
08: 2011_09_30_drive_0028 001100 005170
09: 2011_09_30_drive_0033 000000 001590
10: 2011_09_30_drive_0034 000000 001200

closed 2017-12-20 by AbnerCSZ

robinloujun, 2018-09-03:
Hello, did you manage to convert only the velodyne data? I didn't find option for that (only raw_synced, odom_color and odom_gray). Thanks  a lot in advance!

chubukeji, 2022-01-20:
@AbnerCSZ hello，the 03: 2011_09_26_drive_0067 can't be found in official website. Do you have this dataset? Or，do you know why it can't be found? my Email 2661076951@qq.com

Petros626, 2024-08-20:
@AbnerCSZ did you tried it only with the raw velodyne data or does the velodyne data work too?