tomas789/kitti2bag #18: How to convert odometry_sequences to bag

lynubot, 2018-03-17:
I'm running this package on Ubuntu 16.04 with python aliased to python2.7, I ran into this exception message:
`Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 394, in <module>
    main()
  File "/usr/local/bin/kitti2bag", line 359, in main
    kitti = pykitti.odometry(args.dir, args.sequence)
  File "/usr/local/lib/python2.7/dist-packages/pykitti/odometry.py", line 34, in __init__
    self._load_calib()
  File "/usr/local/lib/python2.7/dist-packages/pykitti/odometry.py", line 172, in _load_calib
    data['T_cam0_velo'] = np.reshape(filedata['Tr'], (3, 4))
KeyError: 'Tr'
`
How to resolve this question

closed 2018-03-21 by lynubot

sunzwdd, 2019-08-08:
I meet the same question. Is there some solutions?

MrMinemeet, 2019-11-12:
More or less having the same problem at the moment.
```
~/src$ kitti2bag -s 00 odom_gray
Traceback (most recent call last):
  File "/home/alex/.local/bin/kitti2bag", line 394, in <module>
    main()
  File "/home/alex/.local/bin/kitti2bag", line 359, in main
    kitti = pykitti.odometry(args.dir, args.sequence)
  File "/home/alex/.local/lib/python2.7/site-packages/pykitti/odometry.py", line 33, in __init__
    self._load_calib()
  File "/home/alex/.local/lib/python2.7/site-packages/pykitti/odometry.py", line 169, in _load_calib
    data['T_cam0_velo'] = np.reshape(filedata['Tr'], (3, 4))
KeyError: 'Tr'
```

valgur, 2019-11-12:
You can try [my personal `kitti2bag` fork/pull request](https://github.com/valgur/kitti2bag), which fixes the breaking changes from newer `pykitti` versions you encountered. The CLI command is slightly different as well: `kitti2bag odom -c gray -s 00`.

MrMinemeet, 2019-11-12:
@valgur I tried your fork you linked to.
Sadly it still gives me the KeyError: 'Tr' 

```
kitti2bag odom -c gray -s 00
Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 11, in <module>
    load_entry_point('kitti2bag==1.5', 'console_scripts', 'kitti2bag')()
  File "/home/alex/.local/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/alex/.local/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/alex/.local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/alex/.local/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/alex/.local/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "build/bdist.linux-x86_64/egg/kitti2bag/__init__.py", line 317, in convert_odom
  File "/home/alex/.local/lib/python2.7/site-packages/pykitti/odometry.py", line 33, in __init__
    self._load_calib()
  File "/home/alex/.local/lib/python2.7/site-packages/pykitti/odometry.py", line 169, in _load_calib
    data['T_cam0_velo'] = np.reshape(filedata['Tr'], (3, 4))
KeyError: 'Tr'
```

valgur, 2019-11-13:
It seems like the odom calibration files you have might simply be broken/empty for some reason? Try extracting and overwriting the [calibration files](https://s3.eu-central-1.amazonaws.com/avg-kitti/data_odometry_calib.zip) again.

MrMinemeet, 2019-11-13:
ok. So I redownloaded the [21GB kitti dataset](http://www.cvlibs.net/download.php?file=data_odometry_gray.zip) tried running kitti2bag which gave me the same error as above . So I replaced the calibration files with the ones that you linked to. When running it with the replaced calibration files it throws me a new error.
```
alex@W550s:~/Desktop/dataset$ kitti2bag odom -c gray -s 00
Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 11, in <module>
    load_entry_point('kitti2bag==1.5', 'console_scripts', 'kitti2bag')()
  File "/home/alex/.local/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/alex/.local/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/alex/.local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/alex/.local/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/alex/.local/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "build/bdist.linux-x86_64/egg/kitti2bag/__init__.py", line 317, in convert_odom
  File "/home/alex/.local/lib/python2.7/site-packages/pykitti/odometry.py", line 35, in __init__
    self._load_poses()
  File "/home/alex/.local/lib/python2.7/site-packages/pykitti/odometry.py", line 228, in _load_poses
    except FileNotFoundError:
NameError: global name 'FileNotFoundError' is not defined
```

valgur, 2019-11-14:
Looks like `pykitti` only supports Python 3. Try using it instead.

MrMinemeet, 2019-11-15:
ok I installed your `kitti2bag` using pip3 now. had to install a few extra things that where missing.
Now I'm stuck with this one:
```
alex@W550s:~$ kitti2bag 
Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
Traceback (most recent call last):
  File "/home/alex/.local/bin/kitti2bag", line 11, in <module>
    load_entry_point('kitti2bag==1.5', 'console_scripts', 'kitti2bag')()
  File "/home/alex/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/alex/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "/home/alex/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/home/alex/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/alex/.local/lib/python3.6/site-packages/kitti2bag/__init__.py", line 21, in <module>
    from tf.transformations import quaternion_from_euler, quaternion_from_matrix
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf/__init__.py", line 30, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)
```

AlterLimbo, 2019-11-16:
I have to say the repository is out of date, while the core developer seldom maintains it .
I think you can amend the source code on your own with just a little effort to make it compatible with latest pykitti. 
I tried [valgur's fork](https://github.com/valgur/kitti2bag) before and it worked well. 
Also, you can try [my fork](https://github.com/AlterLimbo/kitti2bag). Please follow the `How to install it?`  & `How to run it?` instruction to make it work.

MrMinemeet, 2019-11-16:
Nice got it working now. I'm not sure if it's because of the other way to install `pykitti` which is mentioned in @AlterLimbo 's fork or it didn't work as I tried to install kitti2bag using `pip3 install git+https://github.com/valgur/kitti2bag.git'.
Either way both forks work now. 👍
Thank you so much

heethesh, 2019-12-08:
@AlterLimbo's I had issues with `tf_static` not being published. @valgur how can I publish point clouds as well for odometry sequences using your version?

valgur, 2019-12-10:
@heethesh Point cloud conversion for odometry bags is not yet implemented, unfortunately. You can try adapting the relevant part from the raw bag conversion code for odometry bags, if you have a need for this right now. I'll probably add this at some later point.

AlterLimbo, 2019-12-11:
@heethesh Hi, does the generated odometry bag contain `/tf_static`. If it does, the static transformation utilizes `ros_tf2`, which just needs to be pulished once.
You can also load `rviz/kitti.rviz` from my fork to rviz, and play bag to visualize it.