tomas789/kitti2bag #20: command not found

DamonMIN, 2018-05-01:
@tomas789 Dear sir, when I successfully finished all the previous installation procedures and run the command "kitti2bag -t 2011_09_26 -r 0002 raw_synced .".  There is an error "kitti2bag: command not found".  I checked the kitti2bag package with "pip list", which was installed successfully.

ravishk1, 2018-10-05:
I have got the same error. pip successfully installed kitti2bag, but when I tried to call kitti2bag, it showed that   **kitti2bag: command not found**. How to resolve this error?

chhamza2010, 2019-01-30:
try this 
sudo pip install kitti2bag

OmarNsiri, 2019-03-17:
I'm having a similar problem. when I issue: **kitti2bag -t 2011_09_26 -r 0002 raw_synced .**
I get : **Could not load module 'pykitti'. Please run `pip install pykitti`**

jy723, 2019-03-25:
@OmarNsiri hi,I have got the same error.
first,sudo pip install numpy --upgrad, sudo pip install scikit-image.
second,[image]

kitti2bag 2011_09_26 0002.

J2eremy, 2019-03-26:
> @OmarNsiri hi,I have got the same error.
> first,sudo pip install numpy --upgrad, sudo pip install scikit-image.
> second,#10
> kitti2bag 2011_09_26 0002.

I updated the numpy and scikit-image but the problem still exists.
I get : Could not load module 'pykitti'. Please run pip install pykitti.
And how #10 help to solve this problem?
Thanks a lot.

J2eremy, 2019-03-26:
> @OmarNsiri hi,I have got the same error.
> first,sudo pip install numpy --upgrad, sudo pip install scikit-image.
> second,#10
> kitti2bag 2011_09_26 0002.

@jy723 hi,
I updated the numpy and scikit-image but the problem still exists.
I get : Could not load module 'pykitti'. Please run pip install pykitti.
And how #10 help to solve this problem?
Thanks a lot.

mnik17, 2019-06-08:
@J2eremy Hi,

Have you found a solution to that problem? I'm getting the same error, although I tried the proposed solutions and I have updated all dependencies.

valgur, 2019-06-09:
You can replace the 
```python
try:
    import pykitti
except ImportError as e:
    print('Could not load module \'pykitti\'. Please run `pip install pykitti`')
    sys.exit(1)
```
lines with `import pykitti`. That way the exact ImportError message will not be hidden.
The issue is likely to be a missing `cv2` dependency, which pykitti is accidentally trying to import but is not actually used or listed as a dependency either.

mnik17, 2019-06-10:
@valgur 
Thanks for the reply, it turned out you're right. It was in fact a missing cv2.

I'd also like to point out for future reference that if you try installing with python2.7/pip2  you should install an older version of matplotlib, since the current one is requiring at least python3.6.

Gold-finger99, 2020-01-30:
Can you explain the method in detail? I had the same problem, but I was new to it

cicly, 2020-06-21:
first remove the kitti2bag:
```
pip uninstall kitti2bag
```
second install kitti2bag by root:
```
sudo pip install kitti2bag
```

drfg0119, 2020-07-13:
> first remove the kitti2bag:
> 
> ```
> pip uninstall kitti2bag
> ```
> 
> second install kitti2bag by root:
> 
> ```
> sudo pip install kitti2bag
> ```

It works for me! Thanks

SS47816, 2020-11-22:
> first remove the kitti2bag:
> 
> ```
> pip uninstall kitti2bag
> ```
> 
> second install kitti2bag by root:
> 
> ```
> sudo pip install kitti2bag
> ```

Thanks so much! It solved my problem!

nisailhan, 2021-04-28:
> first remove the kitti2bag:
> 
> ```
> pip uninstall kitti2bag
> ```
> 
> second install kitti2bag by root:
> 
> ```
> sudo pip install kitti2bag
> ```

It works for me too. Thanks!

junmeng6025, 2022-07-19:
> first remove the kitti2bag:
> 
> ```
> pip uninstall kitti2bag
> ```
> 
> second install kitti2bag by root:
> 
> ```
> sudo pip install kitti2bag
> ```

Thanks, that worked for me. My problem was #54 . When I reinstall kitti2bag using sudo, it claimed that `bash: /home/jun/anaconda3/envs/kitti/bin/kitti2bag: No such file or directory` , then I run `pip install kitti2bag` again after `sudo pip install kitti2bag`, everything then worked.

Gold-finger99, 2022-07-19:
已收到感谢您得来信

Gold-finger99, 2022-11-17:
已收到感谢您得来信

Rolandxx7, 2023-06-09:
For Ubuntu 20.04,  please just change the python version into 3.8.10, which can solve this problem well.

Gold-finger99, 2023-06-09:
已收到感谢您得来信

Tim-Liu-1, 2023-08-13:
> first remove the kitti2bag:
> 
> ```
> pip uninstall kitti2bag
> ```
> 
> second install kitti2bag by root:
> 
> ```
> sudo pip install kitti2bag
> ```

This method works for me! Thank you so much!

Gold-finger99, 2023-08-13:
已收到感谢您得来信