tomas789/kitti2bag #23: Run the Example

work4it, 2018-09-06:
Hello,
I wanted to try the example. But when I enter the line 
`wget http://kitti.is.tue.mpg.de/kitti/raw_data/2011_09_26_drive_0002/2011_09_26_drive_0002_sync.zip`
I get the following error:
`--2018-09-06 16:29:10--  http://kitti.is.tue.mpg.de/kitti/raw_data/2011_09_26_drive_0002/2011_09_26_drive_0002_sync.zip
Auflösen des Hostnamen »kitti.is.tue.mpg.de (kitti.is.tue.mpg.de)«... fehlgeschlagen: Der Name oder der Dienst ist nicht bekannt.
wget: kann die Host-Adresse »kitti.is.tue.mpg.de« nicht auflösen
`

I think the website is not available. Is there another way to get the sample data?

tomas789, 2018-09-06:
Hi, I can confirm that neither can I download the file. Looks like authors of the KITTI dataset moved the files to the S3. The new address is https://s3.eu-central-1.amazonaws.com/avg-kitti/raw_data/2011_09_26_drive_0002/2011_09_26_drive_0002_sync.zip
I got this from the official KITTI website.

work4it, 2018-09-06:
Thank you for the fast answer. I was able to download and extract the data from the page. However, if I enter the command, this message comes:

>~$ kitti2bag -t 2011_09_26 -r 0002 raw_synced .
Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 12, in <module>
    import tf
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)

Is there any mistake?

work4it, 2018-09-06:
Ok  in #15 there is the solution:
`sudo pip2 install kitti2bag`
Why i need this command?

tomas789, 2018-09-06:
Because ROS only supports (at least officially) python 2. This is not a requirement of the kitti2bag. It works with python 3 too but the ROS-related dependencies are only installed for the python 2. Hope this will change in the future.