tomas789/kitti2bag #27: Hi,

Ghallabi, 2018-11-29:


versatran01, 2018-11-30:
When you say "ring id", does that mean which beam each point belongs to? For example, if one is using VLP64, then there would be 64 different ring ids?

Ghallabi, 2018-12-01:
Yes, exactly. 
Is this label already available ?
Best,

versatran01, 2018-12-02:
I don't know, but I was trying to do the same thing but looks like it is not possible just from the data.

versatran01, 2018-12-03:
One could possibly calculate the altitude of each point and then assign a corresponding ring id?

aidanjones3, 2019-03-24:
I also am interested in determining a solution to this problem. I had two thoughts on the matter: The first being if the kitti datasets somehow have the points ordered already? This may provide some structure to take advantage of. The other would be to look at the timestamp of the points in spherical coordinates per frame. Not entirely sure if this would work but just my thoughts.

geoeo, 2019-10-03:
Formulae is given in https://github.com/tomas789/kitti2bag/issues/49 But it doesnt seem to work. I would love to hear if it works for you

aidanjones3, 2019-10-03:
Will attempt, and update you with results.

versatran01, 2019-10-03:
One (fixable) problem is you won't get clean data. 
I think kitti paper says that they undistorted the point cloud so when you convert them from euclidean to polar coordinates, it will be noisy.

geoeo, 2019-10-03:
Where do they say that? In the dataset paper? Also, does that mean the synced data part should work?

versatran01, 2019-10-04:
Yeah I think in the paper. So raw data should work, not synced because that's where they corrected the lidar data.

geoeo, 2019-10-04:
Thanks for the help. I checked with both and both provide angles which are outside the range of 2-24.8. I looked at the manual again for the HDL 64 E and it states it comes with a calibration.xml. :( Im not really sure how researchers used the kitti velodyne data to provide accurate reconstruction if this is missing

versatran01, 2019-10-04:
Then it's a problem with kitti.

Mr-Matias, 2022-06-08:
Check VincentCheungM/Run_based_segmentation#3 for finding the ring in a non-angle based approach

mr-abramenko, 2023-04-12:
Check https://github.com/Taeyoung96/kitti2bag/commit/a77c6dffe28fc110dca8f7be6167b5b43cfedbff for finding the ring in a angle based approach