tomas789/kitti2bag #37: failed to find match for field 'intensity' 'ring'

DamonMIN, 2019-05-03:
@toams789. I obtained one bag file with the 'kitti2bag'. When this bag file was employed to LOAM_velodyne(https://github.com/laboshinl/loam_velodyne), the program failed with an error "failed to find match for field 'intensity' 'failed to find match for field 'ring'". Is there the intensity and ring field in the sensor_msgs::PointCloud2 in the bag file generated by 'kitti2bag'?

AlterLimbo, 2019-07-26:
Hi, the field 'intensity' is treated as the field 'i'.
You could replace `PointField('i', 12, PointField.FLOAT32, 1)`  with `PointField('intensity', 12, PointField.FLOAT32, 1)` by yourself to fix the problem.
As for the field 'ring', there is no 'ring' info in the origin KITTI bin file. So the bag file don't contain the field 'ring' as velodyne driver does.

HMellor, 2020-11-23:
A ring channel has been added in a custom version of kitti2bag in the LIO-SAM repository. Here is a link to the [relevant chunk of code](https://github.com/TixiaoShan/LIO-SAM/blob/657ef8d6c4245055c7956b04b91ebcdb134eff81/config/doc/kitti2bag/kitti2bag.py#L239).

Although there is a casting issue in the current code which I have fixed in my [pull request](https://github.com/TixiaoShan/LIO-SAM/pull/159) (which may be implemented by the time you read this)

mr-abramenko, 2023-04-12:
- Check https://github.com/Taeyoung96/kitti2bag/commit/a77c6dffe28fc110dca8f7be6167b5b43cfedbff for finding the ring in a angle based approach
- Check https://github.com/VincentCheungM/Run_based_segmentation/issues/3#issuecomment-452686184 for finding the ring in a non-angle based approach