tomas789/kitti2bag #41: Installing from source results in ModuleNotFoundError

thully, 2019-05-16:
I tried to install kitti2bag from the source repository, using the setup.py script. While running "python setup.py install" seemed to work, running kitti2bag after that results in the following error:

Traceback (most recent call last):
  File "/home/thully/anaconda3/bin/kitti2bag", line 11, in <module>
    load_entry_point('kitti2bag==1.5', 'console_scripts', 'kitti2bag')()
  File "/home/thully/anaconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/thully/anaconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/home/thully/anaconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/home/thully/anaconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
ModuleNotFoundError: No module named 'kitti2bag'

I tried this with Python 2 as well as on a system without Anaconda installed, and get the same error. Installing using pip works, though isn't desired as I wanted to work on some potential enhancements to the script (in particular, handling tracklets)

thully, 2019-05-16:
On Python 2 (which is probably what should be used, given ROS doesn't seem to support Python 3), I get the following error instead:

Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 11, in <module>
    load_entry_point('kitti2bag==1.5', 'console_scripts', 'kitti2bag')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 480, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2324, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
ImportError: No module named kitti2bag.__main__

thully, 2019-05-16:
OK - it seems I was installing this the wrong way. It works if I do "sudo pip install -e ." in the directory containing the source. Could this (along with the need to use Python 2) be documented on the wiki?

valgur, 2019-05-16:
You are not installing it wrong. There's currently a minor bug in setup.py - the required `packages=['kitti2bag']` parameter is missing and causes normal installation to fail. Installation in development mode with `python setup.py develop` or `pip install -e .`  still works, somewhat misleadingly.
A fix for this included in #38.

MinaHenein, 2019-12-16:
I could successfully install kitti2bag from source using any of the following commands 
sudo python setup.py develop
sudo python -m pip install -e .
sudo python setup.py install
(I tried them all) but I still get the following error when trying to execute  
kitti2bag -t 2011_09_26 -r 0002 raw_synced .

Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 11, in <module>
    load_entry_point('kitti2bag', 'console_scripts', 'kitti2bag')()
  File "/home/mina/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 476, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/mina/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2699, in load_entry_point
    raise ImportError("Entry point %r not found" % ((group, name),))
ImportError: Entry point ('console_scripts', 'kitti2bag') not found

Any ideas what's happening?
(N.B I'm very unexperienced with pip)

MinaHenein, 2019-12-16:
ok I just got it to work by uninstalling pip and reinstalling it, then uninstalling kitti2bag using pip uninstall kitt2bag, then reinstalling it using pip install kitti2bag, and it works!