tomas789/kitti2bag #42: Version of kitti2bag available with pip mixed up width/height in camera_info

thully, 2019-05-16:
After installing kitti2bag with pip, I noticed that the camera_info width and height are mixed up in the bag file output by the tool. This issue was fixed in the repository, but the latest release (2 years old) doesn't reflect this. Could this be fixed, even just as a one-line 1.5.1 fix?

kdr193, 2020-06-06:
Yeah, it took so much time for me to realize that. Please fix this.

yf26, 2020-09-26:
Just need to swap calib.height and calib.width in line 117 of file kitti2bag.py. Then it works!

sreeharshaparuchur1, 2021-02-06:
Hey, so you can solve this issue by cloning kitti2bag in your catkin/src workspace. Build the ws (preferably using `catkin build`) and source the setup file. Now, navigate to `./catkin_ws/src/kitti2bag/kitti2bag`, you'll find a `__main__.py` script. In that, change the first line from `from  .kitti2bag import run_kitti2bag` to `from kitti2bag import run_kitti2bag`. Change the line in the kitti2bag.py file (swapping height and width, [line number 117](https://github.com/tomas789/kitti2bag/blob/master/kitti2bag/kitti2bag.py#L117)). Now, assuming that you're using the same drive as provided in the example (the raw data -take care that it's not the rectified images as then you'll end up rectifying a rectified image) , have all the required dependencies installed (opencv, scikit-build) and the data (calib and sync) in the correct ordering, run the command: `python2 __main__.py raw_synced -t  2011_09_26 -r 0002`. Ensure that you use python2 and not python3 as ROS runs on python2. Using Python3 may result in errors like `ImportError: dynamic module does not define module export function (PyInit__tf2)`.

Hope this helps