tomas789/kitti2bag #43: No module named kitti2bag.__main__

Yukina1996, 2019-05-20:
kitti2bag is successfully installed after using pip(9.0.1), but when I execute 'kitti2bag -t 2011_09_26 -r 0001 raw_synced .' 
**I got mistakes as belows:**
Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 9, in <module>
    load_entry_point('kitti2bag==1.5', 'console_scripts', 'kitti2bag')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 542, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2569, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2229, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2235, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
ImportError: No module named kitti2bag.__main__

valgur, 2019-05-21:
Looks like the same issue as in #41.

witignite, 2019-11-16:
@valgur, Could you please share/upload some of your `.bag` files generated from KITTI dataset? I would be very appreciated.

I have been tryin to get this work for a few days but not sccessful no matter how hard I tried. I also tried your Docker image but still doesn't work.