tomas789/kitti2bag #44: Pre-build docker image does not work.

yukke42, 2019-06-22:
Thanks for sharing codes.
I pulled the docker image of `tomas789/kitti2bag` then run docker, but the following error occurred.
So, `tomas789/kitti2bag` needs to be updated.

```
$ docker run -v `pwd`:/data -it tomas789/kitti2bag -t 2011_09_26 -r 0002 raw_synced
Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2927, in <module>
    @_call_aside
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2913, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2940, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 637, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 650, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 834, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (numpy 1.11.0 (/usr/lib/python2.7/dist-packages), Requirement.parse('numpy>=1.12.0'), set(['pandas']))
```

emef, 2019-06-28:
You can work around this by upgrading numpy inside the container and then running the command directly:

```
host> docker run -it --entrypoint "/bin/bash" -v `pwd`:/data -it tomas789/kitti2bag

docker> pip install numpy==1.12
docker> kitti2bag -t 2011_09_26 -r 0002 raw_synced
```

valgur, 2019-06-29:
There's a much simpler and working Dockerfile available in [this pull request](https://github.com/tomas789/kitti2bag/pull/38): https://github.com/valgur/kitti2bag/blob/master/Dockerfile

yukke42, 2019-07-03:
@emef @valgur 
Thank you so much. I got working Dockerfile so close this issue.

closed 2019-07-03 by yukke42

OlgaMatykina, 2023-10-20:
> You can work around this by upgrading numpy inside the container and then running the command directly:
> 
> ```
> host: docker run -it --entrypoint "/bin/bash" -v `pwd`:/data -it tomas789/kitti2bag
> 
> > pip install numpy==1.12
> > source "/opt/ros/$ROS_DISTRO/setup.bash"
> > kitti2bag -t 2011_09_26 -r 0002 raw_synced
> ```

Thank you very much! It works!