tomas789/kitti2bag #47: Could not load module 'pykitti'. Although installed.

nilseuropa, 2019-07-27:
```
nils@europa:~/KITTI$ pip show pykitti
Name: pykitti
Version: 0.3.1
Summary: A minimal set of tools for working with the KITTI dataset in Python
Home-page: https://github.com/utiasSTARS/pykitti
Author: Lee Clement
Author-email: lee.clement@robotics.utias.utoronto.ca
License: MIT
Location: /home/nils/.local/lib/python2.7/site-packages
Requires: numpy, pandas, Pillow, matplotlib
nils@europa:~/KITTI$ pip show kitti2bag
Name: kitti2bag
Version: 1.5
Summary: Convert KITTI dataset to ROS bag file the easy way!
Home-page: https://github.com/tomas789/kitti2bag/
Author: Tomas Krejci
Author-email: tomas789@gmail.com
License: UNKNOWN
Location: /home/nils/.local/lib/python2.7/site-packages
Requires: progressbar2, pykitti
nils@europa:~/KITTI$ kitti2bag -t 2011_09_26 -r 0002 raw_synced .
Could not load module 'pykitti'. Please run `pip install pykitti`
```

nilseuropa, 2019-07-27:
Sorry user error.

closed 2019-07-27 by nilseuropa

autoperson, 2019-08-12:
> Sorry user error.

I also has the same problem, how can you solve it?

yujiazhang777, 2019-10-18:
I installed pykitti-0.3.1 but also have this problem.

AlterLimbo, 2019-10-19:
Try to `import pykitti` directly. And you will find out what's the underlying problem.
Usually, it traces back to `numpy`

AlterLimbo, 2019-10-19:
> I installed pykitti-0.3.1 but also have this problem.

@yujiazhang777 

Try to `import pykitti` directly. And you will find out what's the underlying problem.
Usually, it traces back to `numpy`

suyunzzz, 2020-05-13:
> > I installed pykitti-0.3.1 but also have this problem.
> 
> @yujiazhang777
> 
> Try to `import pykitti` directly. And you will find out what's the underlying problem.
> Usually, it traces back to `numpy`

hello,i can import pykitti in Ipython,but these is a issue "Could not load module 'pykitti'. Please run `pip install pykitti " now, could u tell me what should i do?

chongjeelee, 2021-09-27:
maybe only used in python3,not python2

HXB-1997, 2023-09-20:
检查cv2和numpy是否安装 版本是否匹配

ly3106, 2023-12-30:
Reference [here](https://github.com/tomas789/kitti2bag/issues/54#issuecomment-1872481534)