tomas789/kitti2bag #48: About the Imu data

marooncn, 2019-09-03:
Hi, I pay attention to that in [kitti2bag.py](https://github.com/tomas789/kitti2bag/blob/master/kitti2bag/kitti2bag.py),  imu message is assigned as
```
        imu.linear_acceleration.x = oxts.packet.af
        imu.linear_acceleration.y = oxts.packet.al
        imu.linear_acceleration.z = oxts.packet.au
        imu.angular_velocity.x = oxts.packet.wf
        imu.angular_velocity.y = oxts.packet.wl
        imu.angular_velocity.z = oxts.packet.wu
```
But in the definition of kitti raw_data:

> af:    forward acceleration (m/s^2). 
> al:    leftward acceleration (m/s^2). 
> au:    upward acceleration (m/s^2). 
> wf:    angular rate around forward axis (rad/s).
> wl:    angular rate around leftward axis (rad/s). 
> wu:    angular rate around upward axis (rad/s). 

That is the absolute values, but IMU measures the relative acceleration and angular rate in its  coordinate. Why doesn't use the following values?

> ax:    acceleration in x, i.e. in direction of vehicle front (m/s^2).
> ay:    acceleration in y, i.e. in direction of vehicle left (m/s^2). 
> ay:    acceleration in z, i.e. in direction of vehicle top (m/s^2). 
> wx:    angular rate around x (rad/s).  
> wy:    angular rate around y (rad/s). 
> wz:    angular rate around z (rad/s).

marooncn, 2019-09-04:
(ax, ay, az) is the acceleration in the body/vehicle coordinate and (af, al, au) is acceleration in the IMU coordinate.

geoeo, 2019-11-04:
This is fine. It is the correct coordinate system to export. Your second comment refers to gps vs rest.

marooncn, 2019-11-04:
@geoeo sorry, do you mean the original assignment is right？ how to explain my fist comment？

geoeo, 2019-11-04:
I see your point. I think it should be x,y,z as well. Practically however, I tried a motion integration with both and it didnt make any difference. But for completeness it should be x,y,z

marooncn, 2019-11-05:
@geoeo and for my second comment, in the paper it refers both gps and IMU:

> OXTS (GPS/IMU):For each frame, we store30differ-ent GPS/IMU values in a text file: The geographic coordinatesincluding altitude, global orientation, velocities, accelerations,angular  rates,  accuracies  and  satellite  information.  Accelerations and angular rates are both specified using two coordinatesystems, one which is attached to the vehicle body(x,y,z)andone  that  is  mapped  to  the  tangent  plane  of  the  earth  surfaceat  that  location(f,l,u).

uts-hb, 2022-10-28:
> I see your point. I think it should be x,y,z as well. Practically however, I tried a motion integration with both and it didnt make any difference. But for completeness it should be x,y,z

Hi, can I ask you how you got the same trajectory with motion integration using both xyz and flu?  
For FLU what motion integration equation did you used? 
Thank you.