tomas789/kitti2bag #50: Docker hub image is broken

LiyouZhou, 2019-11-26:
Just followed instruction in README and got the following error:

```
docker run -v `pwd`:/data -it tomas789/kitti2bag -t 2011_09_26 -r 0002 raw_synced 
Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2927, in <module>
    @_call_aside
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2913, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2940, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 637, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 650, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 834, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (numpy 1.11.0 (/usr/lib/python2.7/dist-packages), Requirement.parse('numpy>=1.12.0'), set(['pandas']))
```

I don't understand the cause of the error exactly but pandas dropped support for python2 in version 0.24 (so much for semver), and pykitti specifies pandas as a dependency without a version (so much for semver. Semver is dead, long live semver.)

LiyouZhou, 2019-11-26:
https://github.com/tomas789/kitti2bag/pull/51 works around the aforementioned pandas issue.

MABIY, 2021-11-04:
why not fix ?

spencercjh, 2022-05-18:
same issue.

valgur, 2022-05-18:
While not exactly a solution for the broken docker image, you can really make do without docker (or can at least create a much simpler image without a full ROS stack) anyway.
```bash
unset PYTHONPATH; # in case you already have ROS installed and sourced
pip install . opencv-python rosbag roslz4 cv_bridge tf tf2_ros --extra-index-url https://rospypi.github.io/simple/
```