tomas789/kitti2bag #52: How to convert odometry datasets?

zcg3648806, 2019-12-18:
Hi,in Readme introduction,it said "Thanks to @emreay-, this tool can now convert odometry datasets too." but What are the specific instructions for executing this method？Could anyone help me?Thank you.

AlterEtch, 2019-12-20:
Hi, please refer to #18

zcg3648806, 2019-12-23:
@AlterEtch Thank you,but i want to convert odometry images and lidar pointcloud datasets to rosbag.In fact,now I have got the image.bag and velo_pointcloud.bag,but their time are not synchronous,maybe Simultaneous conversion is an effective method, I'm going to give it a try.

AlterEtch, 2019-12-23:
@zcg3648806 If you don't mind, you can try [my fork](https://github.com/AlterLimbo/kitti2bag). Please follow the `How to install it?`  & `How to run it?` instruction to make it work.

zcg3648806, 2020-01-02:
@AlterEtch Thank you, I'll try it. Happy new year!

nedo99, 2022-06-28:
Any follow-up on this one? I tried this fork https://github.com/alterzlw/kitti2bag, but I am still getting:
```
File "/usr/local/lib/python3.8/dist-packages/pykitti-0.3.1-py3.8.egg/pykitti/odometry.py", line 39, in __init__
  File "/usr/local/lib/python3.8/dist-packages/pykitti-0.3.1-py3.8.egg/pykitti/odometry.py", line 175, in _load_calib
KeyError: 'Tr'
```

ulterzlw, 2023-01-03:
Hi @nedo99 try to uninstall pykitti `pip uninstall pykitti` and then install it from source.