tomas789/kitti2bag #54: Could not load module 'pykitti'. Please run `pip install pykitti`

Juuustin, 2020-04-27:
When I tried to do the example, it is said that `Could not load module 'pykitti'. Please run `pip install pykitti`, then I run `pip install pykitti`, it is said that `Requirement already satisfied`.

Any suggestions?
`

valgur, 2020-04-27:
That error message is kind of incorrect and misleading.
Replace [these lines](https://github.com/tomas789/kitti2bag/blob/master/kitti2bag/kitti2bag.py#L6-L10) with just `import pykitti` and try again.

Juuustin, 2020-04-27:
> That error message is kind of incorrect and misleading.
> Replace [these lines](https://github.com/tomas789/kitti2bag/blob/master/kitti2bag/kitti2bag.py#L6-L10) with just `import pykitti` and try again.

Thank you very much! Now it works when I use python2.7 for ROS.
There is another question: Would it be possible to just transform odometry dataset (velodyne laser data, 80 GB) into .bag file?  There are 3 parameters to select: `{raw_synced,odom_color,odom_gray}
`,  but I have no idea which one to choose if I just want to save laser point into .bag.

closed 2020-04-29 by valgur

YangSiri, 2021-03-06:
Hi, @valgur  @Juuustin 

I follow the instructions mentioned above but still can't find pykitti which is already installed in python as shown below.
[image]

[image]

Could you please give me some advice on this? Thanks!

HXB-1997, 2023-09-20:
检查cv2和numpy是否安装 版本是否匹配

ly3106, 2023-12-30:
This problem is caused by `cv2` and `numpy` which are used by `pykitti`. And these [lines](https://github.com/tomas789/kitti2bag/blob/master/kitti2bag/kitti2bag.py#L6-L10) will report this error "Could not load module 'pykitti'. Please run `pip install pykitti`", when `pykitti` try to import `cv2` and `numpy`. However when you install `kitti2bag` in a python3 environment, it is possible that the `numpy`'s version is too high. So, the best solution is to install `kitti2bag` in a python2 environment. The reason likely be that this project was initially developed with python2.

jonarriza96, 2024-02-17:
The entire installation pipeline is broken. Hoping to help any newcomers, the following instructions worked for me:

```
conda create --name kitti38 python=3.8.10
conda activate kitti3
pip install opencv-python pykitti kitti2bag pyyaml rospkg pycryptodomex gnupg
```
and after unzipping the raw data from the instructions
```
kitti2bag -t 2011_09_26 -r 0002 raw_synced .
```

22Xc, 2024-04-22:
> The entire installation pipeline is broken. Hoping to help any newcomers, the following procedure worked for me:
> 
> ```
> conda create --name kitti38 python=3.8.10
> conda activate kitti38
> pip install opencv-python pykitti kitti2bag pyyaml rospkg pycryptodomex gnupg
> ```
> 
> and after unzipping the raw data from the instructions
> 
> ```
> kitti2bag -t 2011_09_26 -r 0002 raw_synced .
> ```

thanks a lot！