tomas789/kitti2bag #55: {raw_synced,odom_color,odom_gray}

Juuustin, 2020-04-28:
Would it be possible to just transform odometry dataset (velodyne laser data, 80 GB) into .bag file? There are 3 parameters to select: {raw_synced,odom_color,odom_gray} , but I have no idea which one to choose if I just want to save laser point into .bag.
Any suggestions?

alteretch, 2020-04-29:
If you don't mind, try [my fork](https://github.com/alteretch/kitti2bag). Please follow the `How to install it?`  & `How to run it?` to make it work.

Juuustin, 2020-04-29:
> If you don't mind, try [my fork](https://github.com/alteretch/kitti2bag). Please follow the `How to install it?` & `How to run it?` to make it work.

I tried your fork, this is an error when I run `kitti2bag -s 04 odom ./dataset/` 
it is said that `kitti2bag: error: argument kitti_type: invalid choice: 'odom' (choose from 'raw_synced', 'odom_color', 'odom_gray')`
So is it possible to select lidar points only?

alteretch, 2020-04-30:
the reason is kitti2bag cmd still links to the old scripts installed by pip. you could try `pip uninstall kitti2bag` or use `/usr/local/bin/kitti2bag` instead. The bag will contain lidar and image both. 
arguments {raw_sync} for raw_sync data, and {odom} for odometry data.

alteretch, 2020-04-30:
if you really want a bag containing only lidar points, use `rosbag filter` to split the generated bag.

closed 2020-05-05 by Juuustin