tomas789/kitti2bag #59: how to run kitti2bag in window?

littlefish111, 2020-11-17:
This seems can only run in linux.
But how to run it in windows?

tanay-o, 2022-08-25:
Did you find anything on this?

Magic-wei, 2022-09-07:
@tanay-o @littlefish111 
I think there are two possible ways to run kitti2bag in Windows:
1. **Run it in Docker**: Windows 10 and 11 can both install the latest Docker Desktop which works on top of WSL2  (I have successfully installed and run Docker Desktop on Windows 10 Home). Make sure you enable and update WSL2 before install Docker Desktop (see [Docker Desktop WSL 2 backend](https://docs.docker.com/desktop/windows/wsl/) and [Download the Linux kernel update package](https://docs.microsoft.com/en-us/windows/wsl/install-manual#step-4---download-the-linux-kernel-update-package) for more details).
2. **Run it in a Python virtual environment** (e.g. via Anaconda): I tested this on Ubuntu but it should works on Windows as well. See #62 for Python dependency installation. BTW, some packages like `tf2-py` and `roslz4` needs to be built from wheels, which needs additional packages to be available. On Ubuntu, I install them by running `apt-get install gcc g++ libboost-all-dev liblz4-dev`. If you can find a way to install them properly on Windows, then I think everything should work well as it does on Ubuntu. I will test this on Windows 10 in the future and update my post here.

Docker looks promising for you guys since Windows 10 and 11 both support WSL2 and thus you can run a Linux container and all the methods to install and run kitti2bag and maybe other applications should work in your Linux container.