tomas789/kitti2bag #60: ValueError: need more than 1 value to unpack

scottymax, 2021-02-25:
Hi,I followed the Readme introduction to convert Kitti datasets,but when I execute 'kitti2bag -t 2011_09_30 -r 0027 raw_synced .',I got mistakes as belows:

Traceback (most recent call last):
  File "/usr/local/bin/kitti2bag", line 394, in <module>
    main()
  File "/usr/local/bin/kitti2bag", line 302, in main
    kitti = pykitti.raw(args.dir, args.date, args.drive)
  File "/usr/local/lib/python2.7/dist-packages/pykitti/raw.py", line 34, in __init__
    self._load_calib()
  File "/usr/local/lib/python2.7/dist-packages/pykitti/raw.py", line 232, in _load_calib
    data['T_velo_imu'] = self._load_calib_rigid('calib_imu_to_velo.txt')
  File "/usr/local/lib/python2.7/dist-packages/pykitti/raw.py", line 147, in _load_calib_rigid
    data = utils.read_calib_file(filepath)
  File "/usr/local/lib/python2.7/dist-packages/pykitti/utils.py", line 74, in read_calib_file
    key, value = line.split(':', 1)
ValueError: need more than 1 value to unpack

Please tell me how to fix it,thank you!

cg-1728, 2022-02-28:
Hello, I met the same problem, how did you solve it?

chaoyanSEU, 2022-05-21:
Hello, I met the same problem, how did you solve it?

XuBin-lgtm, 2022-09-11:
> Hi,I followed the Readme introduction to convert Kitti datasets,but when I execute 'kitti2bag -t 2011_09_30 -r 0027 raw_synced .',I got mistakes as belows:
>
> ```
> Traceback (most recent call last): [...]

Hello, I met the same problem, and I successfully solved it. This error is mainly due to the incorrect file path for executing the generated rosbag statement. For example, you need to run the rosbag statement in 2011_10_03 execute the statement to generate rosbag in this folder.