tomas789/kitti2bag #62: run sample error

chongjeelee, 2021-09-27:
when i run it ,error occured:
 kitti2bag -t 2011_09_26 -r 0002 raw_synced .
Traceback (most recent call last):
  File "/home/lizj/.local/bin/kitti2bag", line 12, in <module>
    import tf
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf/__init__.py", line 30, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)

chongjeelee, 2021-09-28:
i try many times ,failed finally!
but i can convert raw data to bag file with another method ,kitti_to_bag
i suggest change it ,if you have occured the same problem!

Mr-Matias, 2022-05-16:
I had the same error and fixed it by running:
-> pip uninstall kitti2bag
-> pip2.7 install kitti2bag
Then the "kitti2bag -t 2011_09_26 -r 0002 raw_synced ." command ran fine :)

Magic-wei, 2022-07-09:
I just fixed some bugs and ran `kitti2bag -t 2011_09_26 -r 0005 raw_synced .` successfully.

Solution: Create a new Python 3 virtual environment (I'm using Python 3.9) and run

```shell
# in case you already have ROS installed and sourced
unset PYTHONPATH 

# Install OpenCV Python
pip install opencv-python

# Install ROS Python packages
pip install --extra-index-url https://rospypi.github.io/simple/ rosbag rospy tf tf2_ros tf2_msgs actionlib_msgs geometry_msgs sensor_msgs cv_bridge roslz4
```

@chongjeelee @matigolias

ajay1606, 2022-09-07:
@Magic-wei After running the command still getting error like the below; 

[image]

Any suggestions, please !!

Python Version: 3.9
Ubuntu 18.04 ROS Melodic

Magic-wei, 2022-09-07:
Hi @ajay1606, I tested on Ubuntu 16.04 LTS and Python 3.9 in an Anaconda environment. Every time I works with kitti2bag, I will first run `unset PYTHONPATH` (to remove ROS Kinetic path which use Python 2.7 instead) and everything works well.

So I think there are two things for you to double check:
1. Did you execute `unset PYTHONPATH` before you run kitti2bag? This command will get rid of the influence from ROS Python environment since it is Python 2.7 for ROS Kinetic and Melodic but we need Python 3.x for kitti2bag (since pykitti needs Python 3).
2. Can you check `tf` version by `pip list | grep tf` on your machine? Make sure don't `pip install tf` directly since it will install a wrong package `tf=1.0.0` which is not the `tf` package we need for ROS and thus get errors. I have `tf                 1.12.1.post3` installed on my machine.

ajay1606, 2022-09-07:
@Magic-wei Thank you so much for your quick response. by the way after the following command how can we install `kitti2bag`  package?

```
# in case you already have ROS installed and sourced
unset PYTHONPATH 

# Install OpenCV Python
pip install opencv-python

# Install ROS Python packages
pip install --extra-index-url https://rospypi.github.io/simple/ rosbag rospy tf tf2_ros tf2_msgs actionlib_msgs geometry_msgs sensor_msgs cv_bridge roslz4
```
Should i run this command ?

`pip install kitti2bag`

ajay1606, 2022-09-07:
@Magic-wei Thank you its working well now

Magic-wei, 2022-09-07:
@ajay1606 Awesome, good to know it works for you. :smile:

zhongncku, 2022-10-29:
@Magic-wei hope u have a good day. I have this problem, could u pls help me! Thank u in advance
https://github.com/tomas789/kitti2bag/issues/68#issue-1428161117